Geektoolkit/Dynaframe3 #101: Docker Server

Protektor-Desura, 2021-03-26:
Instead of having a bunch of different web interfaces to connect to each picture frame in your home. Why not have a main program that they all connect to on a master server with a single web interface to upload everything to a single place, set settings for each one in one web interface to make it easier to manage all your frames in your home? Doing it under docker would also mean that it would be way simpler for most people to setup and they wouldn't have to know what libraries and such to install and it would just work. You would also be in control of all the versions of everything with a docker as well.

Geektoolkit, 2021-03-28:
Yeah, @RichN001 has mentioned the idea of having a webserver as well, and a few people have mentioned the hassel of managing multiple frames.  

I'll look into it. A central place to manage the frames would be powerful, and would allow for much better and easier web development than we're doing right now (plus we'd gain stability).  The toughest thing is the avalonia framework that is giving me the smooth (on a pi4) animations isn't super thrifty with memory, and we're already close to the pi3's limits for memory pressure causing segfaults if you load too large of an image even.  I have some memory improvements I'm working on to help there, and maybe I have a requirement of a pi4 4gb as a 'server' if you're going to manage several or a dedicated pi to run apache. I'll have to see how thrifty apache is and if I can run it behind Dynaframe reliably....on one hand Dynaframe could drop webserver requirements which would be handy regardless. 

It's a great suggestion, I'll look into it and do some experimenting.

dibyadip-das, 2021-04-05:
I am looking for the same solution. You can take advantage of Github Action to build your code and push your docker image.

Protektor-Desura, 2021-04-05:
I would remove the web server setup running on the frames and put a central website on another machine to handle the web interface. The frames can then just talk to the main server to get pictures, configuration and such. I would think it would make for a much smaller setup needed on a Pi running the frames. You don't have images, you don't have a web server running there. It's just basically a client connecting to the server and throwing up the pictures from the server. It would also mean in theory you don't have to duplicate images on each frame/Pi as well since it would just pull them from the server. It is across a LAN so speed shouldn't be an issue for pulling files.

qwksilver, 2021-04-13:
It would be nice to see a pitcher-catcher type setup, that way both sides can be built into the single frame if that frame is going to be completely stand alone, otherwise you install both on a pi4 and can use lesser units like the pi3 to just catch and display the photo stream.

drinehimer, 2021-11-22:
@Geektoolkit I have the blazor migration portion done. In the interest of not making the PR too big I would like to go ahead and do one just for the blazor portion and a seperate for docker. In order to not have to create a 3.0 followed shortly by a 4.0 I think it would be best to create a 3.0 branch that I can merge to until the second PR is complete.

Geektoolkit, 2021-11-23:
Dan, that sounds great. I just added you to the collaborators group so you have access to handle checkins and such. Since you go that involved with the code you should be able to help manage it :)  Feel free to add me to the PRs.  The way I've handled the builds here has been a bit fast and loose because the 'official' releases are all tested by the community and then promoted. So don't feel like you have to get everything rock solid to create a beta release and get feedback.  Please note that the top of tree though has a database change and may have inroduced other playlist bugs.  
  My dynaframe pro fork that I'm planning on releasing as an image release (no source, but better supported) is so far removed from this at this point that I'm not of much help, but that's ok and was always my plan. I want the community to drive this forward and have it as a very hackable setup that they can make solve issues they have.  I appreciate the passion and work and I'm excited to see the results!

drinehimer, 2021-11-24:
Sounds good. I went ahead and created the 3.0 branch and a PR to that. It doesn't effectively change much to the end user with this PR although it does break/improve the endpoints to follow normal REST standards. If you or the community feel like it is worth going ahead and shipping before I get the docker change - might take a couple months depending on how much time the family gives me :) - then I would be happy to do that and get the documentation updated.

In terms of CI/CD I can probably help with that a little down the road as well. I have a lot of experience setting up CI/CD. Primarily Azure DevOps, but the project I have been working on for my company most of this year was done on Github with Github Actions.

Geektoolkit, 2021-11-24:
Cool I should have time to review it tomorrow evening. I appreciate the assistance. I totally get the family aspect!

drinehimer, 2022-06-07:
I was finally able to get this completed. Looks like the first PR is still open. I went ahead and opened a second to a 3.0 branch.

Geektoolkit, 2022-06-07:
Got it, both should be completed. looks like a ton of great work!

drinehimer, 2022-06-08:
I have merged both PRs. It is ready for any additional testing you or others want. I already had it running in combined mode for a week on my living room monitor and then switched to running the server on a k3s cluster. The docker file is set up to build the container, but a registry will be needed to push to.