Geektoolkit/Dynaframe3 #102: WebGUI shutdown triggers reboot

A380Coding, 2021-03-29:
Hitting the shutdown button in the webGUI actually triggeres a reboot command: http://IP:8000/default.htm?COMMAND=REBOOT

Guess it is a copy&paste issue. Reboot and shutdown button both have the reboot command.

Version 2.18

Geektoolkit, 2021-03-29:
Ah yeah, the 2.18 version number was a miss on my part, I may in place update for that. Good catch on the reboot/shutdown, I'll work to get those updated

RichN001, 2021-03-29:
ah nope that one is on me haha... the link on the WebUI Shutdown button is not correct.

Geektoolkit, 2021-03-30:
Sigh. I'd fire you if i paid you. j/k, you're awesome. I should fire me :D  Ok we can get that in for the next update.

Geektoolkit, 2021-03-30:
Ok no worries we can get that in for the next update, along with my boneheaded version update to 2.19

RichN001, 2021-03-30:
def the link for shutdown is reboot, but on the frame im testing on /default.htm?COMMAND=SHUTDOWN is not working.. might have to dig to find that make sure the command is ok. Ill take a look later.

Geektoolkit, 2021-03-31:
I believe it takes a minute (I execute a 'shutdown' vs. a 'shutdown now'.  I'll change it to 'shutdown now' though. i think I had it that way to allow for windows/linux cross platform support, but I'm ok dropping windows support for now unless there's a huge demand for it.

Geektoolkit, 2021-04-01:
This is fixed in 2.19 (and better...it does an immediate shutdown). Please reopen if you see it regress. Gui is also fixed to say 2.19

closed 2021-04-01 by Geektoolkit