Geektoolkit/Dynaframe3 #105: Sync does not stop local

CwnAnnwn, 2021-03-29:
I'm running 2.13 on a pair of Pi4's. One (Tea) is master, and the other (Coffee) is synched to it. Coffee also grabs the slides from a Samba share on Tea. Coffee is switching slides on its own schedule as well as the master schedule.

E.G. Tea tells Coffee to switch to slide 12, it does. Then 10 seconds later Coffee switches to slide 7. In 20 seconds Tea tells it to go to 13, followed by Coffee going to 8 after 10 seconds.

CwnAnnwn, 2021-03-29:
Upgraded to 2.18 (UI says 2.17), same issue.

Geektoolkit, 2021-03-30:
Ah gotcha, please go to the control tab on Coffee and hit 'pause' and see how it behaves then.  i have a workitem to add that automatically along with some other settings such as the transition times so that they sync better. The times need to be synced to an actual clock as the network latency is causing things to not transition as perfectly as it can be. But first I have some logging work to do so we can actually debug better :)

What you're seeing makes snese to me....tea isn't paused I'm betting, so it's timer is running as well, and when it expires (30 seconds by default) then it's transitiong over. Working on making that all better.

qwksilver, 2021-05-28:
This behavior is fine tuned in 2.21 +.  Remote frames should auto pause when receiving image changes from a main frame.

closed 2021-05-28 by qwksilver