Geektoolkit/Dynaframe3 #107: 2.19, does not start until

Archonw, 2021-04-01:
Hi,
i started testing 2.19. At first try it didn't start up right after the update process. I need to remove all of the old dynaframe stuff.
Now it starts normal, and i think it works as aspected.

Geektoolkit, 2021-04-02:
when you say 'didn't startup right' was it the webUI looking all goofed up? or did it not start running the slideshow?  Asking because I saw one issue with the website being goofed up, but am trying to find out if there are more.

Archonw, 2021-04-02:
It was the slideshow.

Geektoolkit, 2021-04-05:
Ok that's wierd. I'll keep an eye out for it when I get to 2.20. Thanks for the report, will leave this open in case others hit it.

qwksilver, 2021-05-10:
i'm encountering this on one unit based on pi3b  and not on my other unit based on pi4

qwksilver, 2021-05-28:
resolved, or not repeatable, on 2.2 +

closed 2021-05-28 by qwksilver