Geektoolkit/Dynaframe3 #108: Not working after installe

onmyownnn, 2021-04-03:
I ran the install command:
sudo wget https://github.com/Geektoolkit/Dynaframe3/releases/download/2.13/install.sh && bash install.sh; rm -f install.sh ; sudo reboot

but after the reboot nothing is popping up. I tried navigating to my IP with the port 8000 and still nothing. 

I am using Raspian installed on a desktop is that helps.

Thank you.

Geektoolkit, 2021-04-03:
Hmm that may be a new install type. Are there any pictures in the pictures folder? Is there a user named pi? The scripts are made for a raspberry pi, a desktop may break that in interesting ways.

If you go to the Dynaframe folder please see if there are logs there. Or if there is a Dynaframe folder at all in the user directory.

You may have to download and unzip the zip file in the release, chmod Dynaframe to be executable and try that. I'll be interested to know where this broke so I know how to make the script more generic

onmyownnn, 2021-04-03:
Thank you for responding Joe. I had originally tried to install on Mint, but was getting errors. I looked through some of your hard work and saw that it kept referencing /home/pi so I attempted to create a user for pi and try that route, but no luck. I then installed raspian desktop and here I am. I check and the logs and it stated that /lib/ld-linux-armhf.so.3 was missing. I created that directory and still no luck

Geektoolkit, 2021-04-03:
Oh!  Is this an x86 machine by chance? or x64 even?  (vs. an arm processor). If so that would explain it, the releases I do are targeting arm.  we'd need a package for x86 to get it running on an x86 machine.

onmyownnn, 2021-04-03:
That makes sense. It is a  x86_64. I was trying to install it on a more powerful machine that I had laying around. I did see in one of your videos that you had it installed on a windows machine so thought maybe it would work on here to. Couldn't find an installer for Windows so installed linux on the machine hoping to get it working. Thank you for helping out. Do you have any plans to release for x86 in the future?

Geektoolkit, 2021-04-04:
I can try to make Linux x86 compiles for major versions and can try to make one for 2.19 in the next day or so.

onmyownnn, 2021-04-04:
Thank you Joe, but you dont have to go to all the trouble. I wasnt sure if it was already available and I just was unaware of where to get it. I'll probably just buy a new Pi so that I can use Dynaframe

qwksilver, 2021-05-28:
Closing this, but will add x86_32 / x86_64 linux future support

closed 2021-05-28 by qwksilver