Geektoolkit/Dynaframe3 #110: Can't figure out how to use the web UI

imanidiot1, 2021-04-10:
I've installed multiple times and the slide show with standard settings does start up after install but how do I get to web ui to control speeds, rotation, etc. I've got it to come up 3 times by opening chromium and inputting the ip address shown on slideshow start up. But every other time I've tried to open I get error messages with connection and proxy settings. Why does it work sometimes and not others or become non functioning while using? The few times I have got the web UI to open I try to change the settings and then I get error messages stating the site can't be reached and try checking connection proxy and firewall. Then I'm lost. this has only worked like twice out of the 40 or so times I've tried.  I also need info on how to pull pictures from a USB vs folders on the sd card. I've tried using the /media/USB as file but then get error messages again. I've formatted the hard drive and reinstalled multiple times I always get the same thing. Nothing but errors so frustrating because this would seem to be perfect for what I'm looking for if I could get to work for my theater room.

Geektoolkit, 2021-04-11:
Thanks for reaching out on here I'm glad to help. The website for it is the up with :8000 after. You should be able to do that from any computer on the same network.

It only will show the website if it's running. I recommend a different computer or a phone (on wifi on the same network). You don't have to reboot or anything for changes, just hit save.

Also there's no need to reinstall the OS. Even if it went bad you can delete the folder and rerun the install script. I don't do anything OS specific to keep things easy and stable.

Please try with that info and see if you can get the website and rotation to work. Also please don't move the mouse on the Dynaframe that will likely crash it.  Known issue that's being worked on.

You can find log files in the Dynaframe/logs directory.

Hopefully that helps. I'll work with you and we can figure this out

allan-landsman, 2021-04-20:
Hello.
I'm encountering what seems to be the same issue.
If I reboot, Dynaframe launches and shows the IP and port.
I press escape (making sure to not touch the mouse) to exit. I try opening the IP:port in the browser and the site can't be reached. Most of the time.
Sometimes rebooting corrects the issue, sometime not.
Sometimes killing the Dynaframe process (pkill Dynaframe) and then launching with ./Dynanframe from the Dynaframe directory works, and then exiting using escape works, sometimes not.
I've had times where nothing other than deleting Dynaframe and reinstalling it works.
I'm running on a Pi 4B / 2GB with the standard Pi desktop.
The only thing on the Pi other than the OS is Dynaframe.
I've tried with 2.13 and 2.16, similar results.

Any and all help would be greatly appreciated.

Geektoolkit, 2021-04-20:
To be clear...the website is hosted by the Dynaframe process...so if you hit escape and kill the process, the website won't be there anymore. The idea is that dynaframe just runs, showing pictures, and hosts the site in the background, and any device on the same network can use the webpage to control it. 

You shouldn't hit escape on the frame for any reason unless you need to manage the linux instance. if the frame is displaying pictures and you can ping the IP, then you should be able on another device on the same network to go to http://{IPaddresshere}:8000 and then get the menu and remotely manage it. This was my solution for people being able to manage it with a phone/tablet/webbrowser  but keeping the traffic local

Let me know if that makes sense and works. If it doesn't, the next thing to do would be to use escape to get to the linux portion, start a console and type in 'ifconfig' to see what IP address it's reporting. then launch dynaframe (./Dynaframe after a pkill works) and then see if you can get to the webpage.

I'll make a video and try to document this better. Hope that makes sense

imanidiot1, 2021-04-20:
Allan this might help before dynaframe opens on bootup you should have a couple seconds to open a web browser. Try that so once it starts another window should pop up with the browser then insert the ip address and the UI should work just fine then close the window when everything is correct and your settings will hold whenever you shutdown and restart. This is what I've been doing.

Geektoolkit from my understanding what your saying is, when dynaframe is running I should be able to open up the ip url on another device and control that way? I have 2 PI running dynaframe with two tvs one needs to be rotated 180 so if I do that would both get rotated because both are using the same ip?

Geektoolkit, 2021-04-22:
Right correct, my idea/design is that you could control it with any web browser on a phone.  If they're on your home network they shouldn't be able to have the same IP (This would require they're joined to your wifi router or wired in using the ethernet port). The router would assign unique IPs, so you can control them seperately.

imanidiot1, 2021-04-22:
Awesome I'll have to check that out using my Android next time it's running. Thanks @Geektoolkit! Your software rocks so far it looks great!

Geektoolkit, 2021-05-15:
Thanks so much, please let me know if you get it working so I can close this out.

imanidiot1, 2021-05-15:
Yes got it working. These look great. Thanks for your work and getting it out so we can enjoy.

imanidiot1, 2021-05-15:
Thanks again

closed 2021-05-15 by imanidiot1