Geektoolkit/Dynaframe3 #113: bad order output

token386, 2021-04-14:
If there is a sequence of images and a video in between, when the video ends, the image that was left just before the video is shown and continues in the next one. ? Could the next one be shown instead of the previous one?

example:
1.jpg 2.jpg 3.mp4 4.jpg ...

You should see the 4 just after the 3 and not the 2, 4

qwksilver, 2021-04-14:
this is actually designed in.
lets say you have a 30 second slide time and a 15 second video. it will call the video in line, but doesn't advance the slide.
at the end of the video it will return to the last slide.  this is so it doesn't skip a photo.  the current recomendation is to curate your videos to be as long or slightly longer than the current slide time.

Geektoolkit, 2021-05-15:
I plan on a feature to toggle this, I've had enough requests for it that it'll be worth doing. There is a PR however it doesn't have the option in to change it which I'd require so I don't break the 'other' behavior which is what it does now. For now please follow quksilvers guidance...essentially you'l lhave to curate the video length if you want that behavior.

token386, 2021-06-09:
thanks , great job