Geektoolkit/Dynaframe3 #117: unable to save settings

daveknoop, 2021-04-29:
Its for my unable to save the settings for Dynaframe3. 
When i go to http://ipadres:8000 and then go to Settings and change some settings. The page says that he is loading after hitting de save button.

By refreshing the page, the changed settings are changed. But by a restart of the pi, the settings are back to default. 

Is there a plays that de settings are changes by editing a file? or is this a bug in this version?

Version: 2.13.0.0 on a RaspberryPi (Raspberry OS)

Geektoolkit, 2021-05-15:
That's odd..the changes get saved as JSON into appsettings.json.  They should persist through reboots or anything. If you refresh the page it loads the settings from there.

Please try this...change a setting, hit save, then close the browser, reopen it, and see if they are there.

qwksilver, 2021-05-28:
@daveknoop has this issue been worked out with a newer version, or some other solution?

dankinchen, 2021-06-08:
@daveknoop I had the same problem with that version, I upgraded to the latest version v2.3 and it is solved.

machinasdei, 2021-11-12:
having the same issue, using v2.19 - i set everything up and click save, and it doesnt save/or persist through reboot.
The web gui is really jacked up too, barely able to use it......

machinasdei, 2021-11-13:
> @daveknoop I had the same problem with that version, I upgraded to the latest version v2.3 and it is solved.

did you mean v2.23? I don't see any release beyond v2.24

machinasdei, 2021-11-14:
> That's odd..the changes get saved as JSON into appsettings.json. They should persist through reboots or anything. If you refresh the page it loads the settings from there.
> 
> Please try this...change a setting, hit save, then close the browser, reopen it, and see if they are there.

hi - I also tried this, didn't work.  also tried multiple versions of dynaframe.  cant get the appsettings.json to populate....it still has the foo bar test stuff in it.  got an example of what a working one looks like?  We simply need it to save our settings and persist after reboot.  thanks in advance sir!

Geektoolkit, 2021-11-15:
I've moved on several versions so I'll have to dig up one from these versions.   It is odd to me that it's not writing it right off, I've not seen or heard of that before. Is this a raspbian install by chance? I usually run it from the pi folder.  Is there anything in the log you can send? there should be a logs folder with logs written there.

machinasdei, 2021-11-16:
Hi, here are all the logs, tons of time trying to figure it out, hopefully it shows there :).   the appsettings.json is default every time i save something....  odd. no changes seen there, but it does it, but when you reboot, all is lost.  the gui is trashed too, doesnt show up right, can send a screenshot of that too;

[dynaframe-2021-11-13-19-50.log](https://github.com/Geektoolkit/Dynaframe3/files/7543008/dynaframe-2021-11-13-19-50.log)
[dynaframe-2021-11-13-20-35.log](https://github.com/Geektoolkit/Dynaframe3/files/7543009/dynaframe-2021-11-13-20-35.log)
[dynaframe-2021-11-13-20-36.log](https://github.com/Geektoolkit/Dynaframe3/files/7543010/dynaframe-2021-11-13-20-36.log)
[dynaframe-2021-11-13-21-10.log](https://github.com/Geektoolkit/Dynaframe3/files/7543011/dynaframe-2021-11-13-21-10.log)
[dynaframe-2021-11-14-08-36.log](https://github.com/Geektoolkit/Dynaframe3/files/7543012/dynaframe-2021-11-14-08-36.log)

[image]

machinasdei, 2021-12-07:
any ideas?

Geektoolkit, 2021-12-07:
Sorry holidays have been crazy. Looking at the logs, I can't see anything obvious. Actually the oddest thing about the logs is how much is working..usually when something goes wrong I can see an error logged. The logs all look how I'd expect them to look on a perfectly working system.

I have no good suggestions other than trying a different SD card, clean raspian buster install, and then installing fresh to see if something got corrupted.  I'd make a folder in the pictures folder named something basic like 'test' and then copy one image into it named something simple like 'test.jpg'. Then start dynaframe up, make sure the webUI shows up, and then hit 'save' to force a settings save.

I've had corruption cause the menu images you're showing...sometimes it happened when I had two browser windows open and they'd compete. Typically I had to remove the Dynaframe folder entirely and install fresh, but I am guessing you've done that as you've moved from version to version.

I wish I had better answers, but at this time I don't. I do have a new version coming out in the coming weeks that I've spent the last 3 months working on that should fix a TON of issues (I had to rewrite large portions of code). It'll be released seperately from this and I should have more info out soon.

Spinner65, 2022-03-01:
Raspberry Pi 4 with Dynaframe 2.19

I had the same symptoms...
-Settings would not persist between reboots
-menu was corrupt as screenshot from @machinasdei 

Deleting  the  .Dynaframe directory and reinstalling solved the issue.

NbrTDB, 2022-04-28:
Same here, Dynaframe 2.19 on a fresh Raspbian 32-bit build, web UI looks exactly as above. Have tried multiple times deleting, re-downloading the install script and re-running.

Odd since before, it *did* work in the exact same device and environment, which was also a vanilla Raspbian build. Kind of regret reflashing the OS since then (I was attempting to fix an unrelated issue which turned out to have nothing to do with software at all).

It's not some weird file permissions issue is it?

NbrTDB, 2022-04-28:
> What do you mean by fresh ? Make sure you are using Buster.

Oof... yep, that's a problem! I meant freshly installed off the Raspberry Pi Imager, which almost certainly isn't going to be Buster, is it? 

Seen the remarks about using Buster before, but I am forgetting that Buster is an older release, now...

> The problem I and others in this thread had was corrupted files and a simple folder delete and reinstall fixed it.

This does not appear to be the case for me, seeing as no amount of deleting and reinstalling resolved things, even across multiple versions.

Though now I'm just confused as to how I managed to have it working perfectly in the first place...? The very first install was done in exactly the same way and that did work perfectly; if I weren't trying to resolve other issues not relating to Dynaframe I would not have reflashed the card.

In any case I guess I'll try again with Buster before all else.

Geektoolkit, 2022-04-28:
Getting this working on Buster is not trivial, as the entire video engine has to be rewritten. I did that for Dynaframe Pro, but I also had rewritten the entire playlist engine and layout engine as well, so I can't easily port it back.  Also I had to rewrite how rotation is done, so even if I ported the VLC portion back, you'd have a system that couldn't rotate.  My answer has been to have people install Buster since that's the easier path, and I don't have bandwidth to backport all of the changes I did for pro. 

If someone in the community decides to take it on I can mentor them and help them with the changes that are necessary, but it's a ton of work to essentially get it running on a new OS that I didn't see provide any other benefits (I didn't see any noticeable perf gains for instance).

Spinner65, 2022-04-28:
> Getting this working on Buster is not trivial, as the entire video engine has to be rewritten.

I assume.you mean Bullseye and bookworm ?
I have it running on buster

 A possible solution to prevent confusion is for the code or the install script to check which version is you are running and throw an error if required. 
Would be handy for both old Dynaframe and the replacement Dynaframe Pro.   You never know then the OS developers may make a change that breaks your code.

Geektoolkit, 2022-04-28:
Sorry yes, updated the comment to reflect bullseye. 
That's a good idea, I'll try to get the detect logic updated on the script to at least warn the user.  For Dynaframe pro I moved to an image based deployment because of issues like this and some others that kept getting hit by the community.  Dynaframe pro has the features and logic for swithcing the wifi to adhoc to allow for pairing without needing to ssh/remote in, but that was even more of a mess to setup properly, so an image based install was the only way to go. That puts alot of work on me and the growing team to put that together and host it, but it results in much less user confusion as they can just image and go.

the script idea is a good one and I'll implement that when I get a chance. Right now I'm working on moving houses so a bit buys for the month as I sell a house, buy a house, pack a house and well...unpack a house :)