Geektoolkit/Dynaframe3 #118: Feature request

elbori77, 2021-04-30:
Hello is possible to add a loop option for video. In case I want to play a video forever. The current behavior the program start loading the same video again going to a loading screen before playing again.

qwksilver, 2021-04-30:
If I am infering correctly you've setup dynaframe with one file  call it screen.mp4 and you want that one file to loop forever after booting.  Can you confirm or describe in more detail?

elbori77, 2021-04-30:
Thank you for reaching out. Yes just like that. One file to loop forever
after booting.

elbori77, 2021-05-01:
Will not be changing the file often.

qwksilver, 2021-05-03:
this it might be much easier to set yourself up a script with a  command line that uses vlc to loop the single video.  the overhead will be lower and the webserver components not needed.  once you have the script, you can simply set it to load on boot.  I'm sure there are other things that might be prudent like turning the system screen saver off, if you plan to use it in portrait you can use omxplayer directly to launch and set the rotation how you want it,  and a watchdog program of some sort that reboots the whole unit on a freeze up or error.

this type of paired down solution would be much less resource intensive, and would loop quite smoothly.

qwksilver, 2021-05-03:
closing issue for loop forever as not practical in project, limited looping items exist elsewhere on git.

closed 2021-05-03 by qwksilver