Geektoolkit/Dynaframe3 #130: Help!

1ln1x, 2021-06-17:
Hi, i encountered some issues
1/ pictures are blown out of proportion 
2/ i am not able to stop dynaframe - tried moving the mouse, but it is stuck at the same picture
3/ http://192.168.xx.xx:8000 isnt working for me ( accessing it via chrome; MBPro15"2014)

so far, i'm using
RPI4B 8gb 
Samsung 27"
did not switch on VNC 
tried the copy-all-photos-into-pictures-via-USB-method.
using 2.12 release ( how to uninstall and update to the latest version?)

thanks in advanced!

--what i did/ attempts log:--
1. followed the instructions to the T (youtube); 
2. copy pasted the "sudo wget https://github.com/Geektoolkit/Dynaframe3/releases/download/2.12/install.sh && bash install.sh; rm -f install.sh ; sudo reboot" 
it ran some scripts, and rebooted my RPI,  and the slideshows began ( yeahh. but it was a short-lived hurray ) 
3somehow it ?crashed out to the desktop after i moved the mouse (unintentionally). then i killed dynaframe with "sudo pkill Dynaframe". however, images still kept popping out randomly.
4. i restarted it by unplugging my power source to my RPI (wince)
second time around, i  disconnected my keyboard& mouse. after restarting, my desktop was visible for a few seconds, and dynaframe started to reboot. showing the address etc. tried accessing the address but had this error returned ->( "This site can’t be reached192.168.1.xx took too long to respond.) Pictures were blown out of proportion, and when i attempted to reconnect my mouse& keyboard, everything froze.
5.tried unplugging and restarting again, but it seems to be the same. 

Am i doing something wrong?
Thanks in advanced!

qwksilver, 2021-06-19:
Based on the above you installed 2.12, did you ever run, "sudo wget https://github.com/Geektoolkit/Dynaframe3/releases/download/2.23/install.sh && bash install.sh; rm -f install.sh ; sudo reboot" 

i recommend vnc, and escape is the correct way to exit dynaframe, if it gets stuck you can connect with putty(ssh) and send a pkill Dynaframe  (*note the capitol D is required)

for #1, getting to settings is the way to set your photo desires under image scaling.
for #3, windows set to not connect, or public networking, vpn set to not allow local connections, things like that are the likely culprit .  i will note that mine connects as http and not https  so checing that, and settings that say don't allow non-https connections might also cause issues.  

look into those things and get back with us

1ln1x, 2021-06-22:
Hi there, thanks for your reply.

What i've done so far,
1/ **removed all of Dynaframe and re-installed the whole package** via :  "sudo wget https://github.com/Geektoolkit/Dynaframe3/releases/download/2.23/install.sh && bash install.sh; rm -f install.sh ; sudo reboot" ---> images ran on boot (hurray), still out of scale ( sigh ), and was able to escape via the esc button ( ^-^ )
2/ checked all settings (MBPro, and also tried with Win10) (made sure SSH and VNC are open ; confirmed with " sudo systemctl status sshd" ; pinged google; double checked my IP address via ifconfig and hostname -I ) but ..... i am still not able to connect via HTTP://xxx.xxx.xxx.xx:8000 ...which i think is my biggest problem right now.

qwksilver, 2021-06-22:
ok if the photos are running then so is the webserver, if the photos stop or are stopped then the web interface is also down.
i use an program called advanced ip scanner to id units on my network, you might try that with the pi off to see if something else is on the same ip.  run it again with the frame running to see if it's not on the ip you think.

mine shows up at http://192.168.0.226:8000/  but it's different for everyone.
pinging the 192.168.0.226 for me gets an answerback (with or without the frame displaying photos)

if you don't mind sharing the actual ip that may provide some insight here, also a pro tip is i've assigned a static address in my routers. 

finally are you connecting via hardwire or wifi, and if it's wifi can you connect to vnc?

1ln1x, 2021-06-22:
Hi there.
All my units are on the same wifi.
I've installed nmap "sudo apt install nmap" and has mapped out the entire server via "nmap -sn 192.168.1.0/24" & sudo "nmap -sn 192.168.1.0/24". So far i'm able see all other connections. 
Now that you've mentioned, Just tried and I am unable to remotely connect to this pi via SSH and DNS!! (gosh, i feel like i'm opening up another can of worms  ) :  
Now i'm thinking.. maybe the last resort is to format my whole microsd and reflash the whole thing - which i am a wee bit reluctant to do it. Do you have any other suggestions/alternatives?

1ln1x, 2021-06-24:
Hi there, a quick update,
i've managed to connect to http://ip:8000 !!! ( it was pure joy to see those pictures up and running!)
It turns out, somehow was that, my modem had a 2.4G and a 5G connection. Once i put both (mac and RP) (Win10 and RP) on 5G, it ran flawlessly (not sure why; but i am super glad it turned out great.)

[Picture below - Interface was flawless and great! ( \**Thank you for the great work  @qwksilver **   !)]
[image]   

**Finally, another quick question, does dynaframe support HEIC format? ( pictures taken with iphone sometimes are saved as HEIC rather than JPEG)**

qwksilver, 2021-06-27:
Excellent! Glad to see you were able to get going.
Currently images are displayed by Avalonia, and based on an answer [here](https://github.com/AvaloniaUI/Avalonia/discussions/5436) it might be feasible to support soon.  I'll pass it on to development threads.

closed 2021-06-27 by qwksilver