Geektoolkit/Dynaframe3 #131: No Dynaframe above 2.19 with Raspberry 3

Archonw, 2021-06-20:
Hallo, 
With my Raspberry Pi 3 I'm not able to get the Dynaframe working above version 2.19. Is there anyone who has the latest version 2.23 running  on a pi 3?

Geektoolkit, 2021-06-20:
I haven't tried it in a pi3 but those builds are really rough. Normally the builds bring in bugs but those will crash or just get stuck. A lot is being rewritten (in a good way).

What are you seeing when you try it?

Archonw, 2021-06-20:
I get a segfault error.

Geektoolkit, 2021-06-20:
Ah can you try a smaller playlist or maybe smaller images (ensure 1080p). I'm working on a sql lite database and a perf improvement but either of those may be causing it

Archonw, 2021-06-21:
With a smaller playlist Dynaframe started. Thank you for that tip.
What doesn't work: - i can't load my big playlist 
- no changes on the infobar settings
- no shuffle

Geektoolkit, 2021-06-27:
the memory management got out of hand as features came in.  I'm moving to a database to trade memory for disk space which should give a great performance tradeoff for load times, and also fix things. This is a big change/rewrite though so please be patient (that's why I'm keeping 2.13 as official release, and 2.18 as unofficial 'safe' release.).  Once this work is done though it'll open up the next generation of work.  Typical devleopment cycle...add features, fix stuff that's broken, fix performance regressions from features, rewrite part of the engine (this is where I'm at) and then stabilize for release. Rinse, Cofee, Repeat :)

Geektoolkit, 2021-09-01:
Archon, I'm going to close this. I'm going to move Dynaframe back to 2.18, the last stable build, and revert the database changes.  I'm planning on forking the project into a "Dynaframe pro" and I'll continue things via that. I'll do an update when I officially launch that project when it's ready (that's what I've been spending my time on)

closed 2021-09-01 by Geektoolkit