Geektoolkit/Dynaframe3 #132: Compile for x64 ubuntu

n4tur3502, 2021-06-22:
I am extremely sorry if this has been asked before but how would one go about compiling this for use on an old laptop running ubuntu?

qwksilver, 2021-06-22:
Sorry, wrong button.  As of now the point releases are for ARM archetecture only.  There may be a few extra builds available for the next stable release, but I am not certain of any final plans.  I would also have you check the closed issues as well, this question has poped up before and there may be a few tips scattered here and there on building for x86 and x64 architecture.

n4tur3502, 2021-06-22:
By some random miracle I managed to get this compiled for my windows pc lol. Now to figure out how to get it working on my linux laptop. Slowly but surely getting there. Not too bad considering this is my first attempt at compiling something. This is fun!

n4tur3502, 2021-06-24:
Well I have tried to get this working for ubuntu x64 and I am over my head on this. If anyone could help me get it working that would be cool as I cant get a pi and I have more old laptops than I know what to do with. I really dont know what I am doing with building/publishing and stumbled onto the publish options and selected linux x64 under target run time. I have no idea what configuration to use or anything. I dont even know how to get rid of the SkiaSharp error. Sorry if this is a lot of help requests, I am learning all this on my own.

OMMJAI, 2021-06-24:
I would also like to know if compiling for ubuntu is possible. I plan to convert my old laptop into a digital photo frame. Along with adding Alexa functionality using the amazon sdk. Would really appreciate if someone could compile for x64.

Geektoolkit, 2021-06-27:
Sorry was out on vacation. Compiling for Ubuntu is possible.  It's currently a visual studio 2019 solution, setup with Nuget to pull in dependancies. If you load it in VS2019 and build it successfully, then you're 90% of the way there. The next steps would be to right click the solution, go to Publish, and then publish it selecting x86/x64 (right now the builds are arm based). Then you'd copy the output to your laptop, sudo chmod +x Dynaframe to make it executable, and then you can launch it with ./Dynaframe.

I plan on making an ubuntu 32bit release as well as a windows release when I do the next official release. However right now it's deep in the next cycle of development as I transition to a database to enable the next series of features (and to fix some awful performance issues it hit).  this means you can try to compile and get it going, or give me a few weeks to get the codebase stable again so I can do an official release.  If you stick with it we'll get you going.  You'll have to do some installs yourself for Ubuntu, and I'm not 100% sure every feature will work (for instance the screen on/off for instance).  But it should work for the 99% case in that configuration, and likely run really well.

OMMJAI, 2021-06-27:
Hi just had a very wild idea of running Dynaframe on rasbian x86.
Is it possible because it is the same operating system? Or will it still have to be recompiled

n4tur3502, 2021-06-27:
I really appreciate the responses. I had actually tried those steps and keep getting the unhandled exception i mentioned above. I tried to find out if I was missing something on linux to make it run but keep hitting dead ends. I wasnt sure if it was related to the package version error thing. I am still learning and this is a really neat project , plus at the end I get a really cool digital frame.

Geektoolkit, 2021-06-27:
re: Raspbian - It won't work because the released builds are compiled for arm. You have to match the OS as well as the architecture (arm/x86/x64).  Though X64 devices will usually run x86, they do not run arm.

For the issue you list above, that looks like a warning, not an error, so you may be ok actually. If you're building on a windows box, I'd try to launch it and see if it launches, (hit f5 and see if it starts). If so then all you have to do from there is publish it to linux/x86

n4tur3502, 2021-06-27:
When I hit f5 it starts giving exceptions and stuff. Ill probably just wait till you get the new stuff done and try again. I did manage to get a version working on my Windows 10 pc, which is where VS is. So I really dont know if im missing something. If you decide to do a version that works in linux on a normal laptop then I'll have like 6 frames for my house lol. All good either way.

Geektoolkit, 2021-06-27:
Oh make sure to run Visual studio elevated. That'll cause it to crash

n4tur3502, 2021-06-27:
Ok cool, running elevated allowed it to run from f5. But no matter what I do i still get the unhandled exception when running on ubuntu x64. I made sure to install the .net 5.0 runtime on ubuntu. When I go to publish it only shows linux x64 and linux-arm for linux options. I select linux-x64, leave the framework to net5.0, config debug (also tried release).  I assume I am not doing something correctly.

Geektoolkit, 2021-06-27:
Ah you're close! Please check install.ah...it apt gets some packages. Maybe one of them is missing on Ubuntu vs raspbian. Also see if you get a log folder in the Dynaframe directory. I publish with the .net framework included so that others don't have to install it. This is important because you're at the same point you'd be if I published it...we have to figure out what's missing or crashing

n4tur3502, 2021-06-28:
Not seeing a log folder when I run it on ubuntu. I installed unclutter from the install.sh file but that didnt seem to help.

n4tur3502, 2021-06-29:
Not sure what I am missing on this still but I do believe it is time for me to throw in the towel. No matter what I do it still gives the same error when running on ubuntu. 

"Unhandled exception. System.ArgumentException: Unknown pixel format: Rgb888x
at Avalonia.Skia.SkiaSharpExtensions.ToPixelFormat(SKColorType fmt) in //src/Skia/Avalonia.Skia/SkiaSharpExtensions.cs:line 138
at Avalonia.Skia.PlatformRenderInterface..ctor(ISkiaGpu skiaGpu, Nullable1 maxResourceBytes) in /_/src/Skia/Avalonia.Skia/PlatformRenderInterface.cs:line 27 at Avalonia.Skia.SkiaPlatform.Initialize(SkiaOptions options) in /_/src/Skia/Avalonia.Skia/SkiaPlatform.cs:line 20 at Avalonia.SkiaApplicationExtensions.<>c__01.b__0_0() in //src/Skia/Avalonia.Skia/SkiaApplicationExtensions.cs:line 20
at Avalonia.Controls.AppBuilderBase1.Setup() in /_/src/Avalonia.Controls/AppBuilderBase.cs:line 304 at Avalonia.Controls.AppBuilderBase1.SetupWithLifetime(IApplicationLifetime lifetime) in //src/Avalonia.Controls/AppBuilderBase.cs:line 179
at Avalonia.ClassicDesktopStyleApplicationLifetimeExtensions.StartWithClassicDesktopLifetime[T](T builder, String[] args, ShutdownMode shutdownMode) in //src/Avalonia.Controls/ApplicationLifetimes/ClassicDesktopStyleApplicationLifetime.cs:line 147
at Dynaframe3.Program.Main(String[] args) in C:\Users\Eric\Source\Repos\Dynaframe3\Program.cs:line 18
Aborted (core dumped)"

Seems I just dont know enough to make it work.

Geektoolkit, 2021-09-01:
I've decided to not go this route. I may make it in the future on my future dynaframe project, but for this one it's going to be too much for me myself to do it. closing this out

closed 2021-09-01 by Geektoolkit