Geektoolkit/Dynaframe3 #139: No sound on 3.5 mm jack audio outlet

mrogojanu, 2021-07-25:
Hi,

First of all thank you for sharing your work !  Great project !
 
I made a DIY photoframe using an old monitor and a raspberry pi 3b+.  Everything works out of the box with small adjustments on  GPU memory with increase to 256 MB for getting smooth video output.   The single problem  is the lack of sound , as audio was configured for 3.5 mm jack outlet.  I assume  Dynaframe  default config is made for HDMI audio outlet.   As a result  kindly request for a solution,  as already glued everything and cannot switch to HDMI audio outlet.

Geektoolkit, 2021-07-26:
If you hook up a keyboard/mouse to it, you should be able to hit 'esc' to stop the software, and then start a terminal and go to raspi-config and change the audio settings.
If you  have ssh or VNC access you should be able to do the same. 

Hope that helps!

mrogojanu, 2021-07-26:
Thanks for quick reply.  Finally I resolved  by  editing the /boot/config.txt with option  hdmi_ignore_edid_audio=1, which pretends that all audio formats are unspuported by display so ALSA will default to analog.  The topic can be closed.

closed 2021-07-26 by mrogojanu