Geektoolkit/Dynaframe3 #143: Dynaframe Always Crashing At/Near End of Directory

allan-landsman, 2021-08-23:
Dynaframe always crashes either a few images before last image /last image/ a few images after last image.
Originally I though it might have been a corrupted image file or the quantity (target ideal is ~300 images + 50 videos) of files, but even with ~40 files (each a 1-3 MB photo) it will crash.
Pi OS stops outputting video, but going to keyboard attached to Pi and hitting Esc drops to Pi desktop.
Rebooting Pi give same/similar result as the initial run.

Not sure what if any logging would be helpful, will happily supply if you can give instructions on what/how.

Running a Pi4 /4GB, vanilla OS updated as of today (nothing else installed on OS other than Dynaframe and what Dynaframe installs).
Running Dyanframe 2.23 (screen states 2.22, but definitely ran install for 2.23).

Geektoolkit, 2021-08-23:
Thanks for the report. 2.2x is a very experimental series where I'm moving to a database format. Please try 2.18 for the latest stable that I'm aware of.

Geektoolkit, 2021-11-07:
Cosing due to inactivity, if you repro this on 2.19 please reopen

closed 2021-11-07 by Geektoolkit