Geektoolkit/Dynaframe3 #144: Settings and Web GUI

mirage1usa, 2021-08-31:
Scenario 1:
I installed Dynaframe3 **2.13** on a Pi3b+ with Raspbian.  Added some images to the Photos folder and rebooted.  Displays images great, but the IP address only showed for one second and images display for 30 seconds.

Went into the settings page on port 8000 and it displayed. Made settings changes to display images longer and they never went into affect. Tried multiple times.

Removed Dynaframe3 **2.13** and installed Dynaframe3 **2.17.**  Images displayed, IP address flew by again and changes on the settings page still do not affect Dynaframe.

Scenario 2:
I then wiped the SD card and installed a fresh ISO of **Raspbian (version 10, buster).** Ran updates, changed the host name, changed the pi password, enabled ssh, and installed Dynaframe 3 **2.17**.  Images display, IP address pops up for a second and the settings page shows a, "This site can’t be reached, 192.168.5.xx refused to connect."

Is there something in the **2.13** version that configures the local web server correctly and freshly loaded, later versions do not?  I seem to be battling multiple issues and wondering if others have experienced them.  

[dynaframe-2021-08-30-20-36.log](https://github.com/Geektoolkit/Dynaframe3/files/7081155/dynaframe-2021-08-30-20-36.log)

Geektoolkit, 2021-08-31:
Hmm interesting. Just to be sure...when you changed the duration, did you hit save at the bottom of the page? I think it is scrolled off the bottom sometimes.  If so, when you check the log you should see the updated Duration displayed. I see a duration of 30 seconds in your log, but I didn't see an update that usually happens after a save is hit.  

The web server is just a listener built into Dynaframe, so dynaframe must be running for it to work.  If you exit the program you'll likely see the 'refused to connect' issue. 

Let me know if either of those suggestions are helpful.

mirage1usa, 2021-08-31:
That is what I was missing!  I expected the web server to be active all the time for adjusting settings. So I pressed ESC to exit the photo stream to make changes.

I was able to access the Dynaframe web server and edit settings.  It is working correctly. It was a misunderstanding of how the interface is expected to be used.

This time I quickly started a web browser session after a reboot, let Dynaframe start displaying images, and then ALT-TAB to the browser to access the settings page (http;//LocalIP:8000).  If there is another recommended method for locally getting to the settings page after Dynaframe is running, I think that would be helpful to add to your Getting Started page.  I just realized that when I SSH'd into the system, Dynaframe was not displaying and could easily launch a web browser. Guess that is the expected usage.

Thanks, glad it was just a user issue!  I have been following for a while now and really appreciate all the time you have put into this.

mirage1usa, 2021-08-31:
Just noticed the closed issue #110 was this exact same issue.  Sorry. I only checked open issues.

closed 2021-08-31 by mirage1usa

Geektoolkit, 2021-09-01:
Hey no worries glad you got it going! Please try to use a web browser from a phone; tablet, pc etc to control the frame. The idea is that it's "remote" controlled :)