Geektoolkit/Dynaframe3 #148: video playing not working (wmv & mp4) on 2.23

Mkka710, 2021-10-04:
Hey! That+s project is so amazing - but it dosn´t work on my pi correctly :-(
I got the same problem as #44 video playing not working.
And I have a second beginner-question: How can I start the programm? I used IP-Adress:8000.

I used version 2.23
System: Raspberry Pi 4
File format: .mp4 and wmv

Many thanks for your help in advance.

Mkka710, 2021-10-31:
[dynaframe.log](https://github.com/Geektoolkit/Dynaframe3/files/7449168/dynaframe.log)

I hope that´s the right logs?
At the moment I used the storage space of  the raspberry. Not a usb stick. 
[dynaframe-2021-10-23-22-50.log](https://github.com/Geektoolkit/Dynaframe3/files/7449169/dynaframe-2021-10-23-22-50.log)
[dynaframe-2021-10-23-22-55.log](https://github.com/Geektoolkit/Dynaframe3/files/7449170/dynaframe-2021-10-23-22-55.log)
[dynaframe-2021-10-23-23-24.log](https://github.com/Geektoolkit/Dynaframe3/files/7449171/dynaframe-2021-10-23-23-24.log)
[dynaframe-2021-10-23-23-36.log](https://github.com/Geektoolkit/Dynaframe3/files/7449172/dynaframe-2021-10-23-23-36.log)
[dynaframe-2021-10-31-16-19.log](https://github.com/Geektoolkit/Dynaframe3/files/7449173/dynaframe-2021-10-31-16-19.log)
[run.sh.log](https://github.com/Geektoolkit/Dynaframe3/files/7449174/run.sh.log)

Geektoolkit, 2021-11-07:
Ok from the logs, it looks like the video is not supported by OMXplayer if I'm reading things right. I'd try to play the video with omxplayer on the pi outside of dynaframe, if it can't play then you'd have to reformat the video file as a file that omxplayer can play.  I see Dynaframe passing it to omxplayer in the log, so not sure what's going on beyond that.

closed 2021-11-07 by Geektoolkit