Geektoolkit/Dynaframe3 #155: Unable to access dynaframe web page

frankups1, 2021-11-22:
I installed dynaframe on a new install of raspbian on a raspberry pi using the first method on the project page. it rebooted with the web url but when I go to the web address it says this site can't be reached

scotthibbs, 2021-11-22:
I might be able to help with this but more info is needed. 1. What was the webpage url? 2. Are the computer and the raspberry pi both on the same network? 3. Keep in mind that the ip address of the raspberry pi needs to be appended with :8000 for example, my Dynaframe is reached with http://192.168.254.233:8000/ in the address bar. Does this help?

Geektoolkit, 2021-11-23:
Scott, just want to say hugely appreciate you jumping in and helping out here. Frank Scott's questions should help us narrow things down