Geektoolkit/Dynaframe3 #156: web page out of sync

scotthibbs, 2021-11-22:
LOVE THIS PROJECT!! Going to do a video of how to make a digital frame with an old monitor with a pi to display QSL cards in my Amateur Radio Station. (I'm able to scan these post cards directly to the digital frame. Love it!) I then go to the webpage to update the file list or wait for the next reboot to have those pics added.

Small issue, when I connect to the digital frame via (http://192.168.254.233:8000/) it can show buttons out of sync. Meaning that the red pause button shows even though the frame is playing. Or shuffling when it isn't etc. I think I've done this when I duplicated the tab in chrome or controlled from multiple computers. At least that is my fix: Duplicate the tab, hit the pause button in the second tab (so that it shows playing) then close the tab. Then the first tab is on track.

The fix may be a query when the webpage is loaded for the button status to display. 
Hope all is well, let me know if you need any testing. 
Scott KD4SIR
scotthibbs@gmail.com

scotthibbs, 2021-11-22:
Forgot to mention I'm using 2.19

Geektoolkit, 2021-11-23:
Yeah, there are alot of website quirk bugs in the way things are handled, and honestly the code for the website is a mess and needed to be rewrittten.  Two options:
  Dan has taken on porting the web code to a proper webserver.  This will help beautify things, and get them into sync. He's working on a PR to make this happen, though it'll take some time.
   Second option: This project has taken quite a bit of time for me, and grown in a way I didn't expect. I plan on doing a patreon of $5  a month for monthly releases of a more supported model that I've been working on for about 3-4 months now through my evenings.  This version includes a redone web page and a bunch of other community fixes.  You'd only have to sign up for one month, but continued support would help me with other features (for instance I'm working on a web hosted site for content so people can easily download content to the frame).  I expect that release and Patreon to be announced on youtube (geektookit) in the next few weeks.  The codebase is So much nicer for me to work on and is supportable. 
     Either way I'm super glad you like it, and I'm  subscribed to your channel and will help in any way I can!