Geektoolkit/Dynaframe3 #158: image size limit?

MasterSergeantD, 2021-12-01:
Is there one?  I have images up to 10MB and it seems like the player blows out when trying to access these files.  Works with image sizes of less than 1 MB, 188 images so far.  Using Pi 4 with bullseye OS, 16GB SD.

MasterSergeantD, 2021-12-01:
I am an idiot.

closed 2021-12-01 by MasterSergeantD

Geektoolkit, 2021-12-02:
I've worked for Master Sergeants, and none of them were idiots, but only deserving of respect for their service and achievements in life :)

I'm glad you got things up and running! I haven't even gotten to Bullseye yet! 
I'm working on a new version that will be called Dynaframe Pro which will be an image based install. It will help solve issues like this so there's less gotchas and an even easier install.  Basically write the image and call it a day.  Hoping to have an announce video on my youtube channel soon!

MasterSergeantD, 2021-12-02:
Thanks Joe!  I am looking forward to it.  Appreciate your service sir!

Tom Davidson