Geektoolkit/Dynaframe3 #163: Videos not playing

PhotonGrabber, 2021-12-15:
I noticed that this has been reported before and it looks like no progress was made.  The issue was closed with no action.

It appears the problem is that the latest version of the OS "Bullseye" has deprecated support for omxplayer.  Omxplayer dies when launched on a fresh install of the OS with a missing shared library for OpenGL.

I've found multiple complaints by various users that omxplayer isn't working and the suggested solution by debian devs is to switch to VLC.

Geektoolkit, 2021-12-15:
There were seperate issues for videos not playing before Bullseye was released due to other issues. However this is indeed a new one, and a bad one...I and qwksilver are seeing this as well and came to the same conclusion...omxplayer is deprecated and the video engine will need to be rewritten to support bullseye.  I'm dissapointed in that decision...alot of projects as you mentioned got blind sided.

The problem with VLC is the last time I tried it, it didn't support rotation on the Pi (It'd just fail silently or show a black screen).  I'll have to see if that was fixed.  I'm investigating moving to ffplay, but that change will take time.  For now the recommendation is to stay on previous builds for this project if you need video support.

qwksilver, 2021-12-16:
To Clarify, the previous (buster) build of PI-OS, and the most stable (for your needs) of the releases here. 2.19 being the last "release" designated version, but there are some advantages in the beta's beyond that.  We'll know more about a release date of improvements to Dynaframe after the new year, Happy Holidays All.