Geektoolkit/Dynaframe3 #166: How to disable "slideshow duration" for videos?

nanpuhaha, 2021-12-21:
_using Dynaframe v2.19 on buster rasbian_

I want to play each videos until the ends, but it play videos only first 30 seconds and skip to next.
The 30 seconds looks like `Slideshow duration between slides (ms)`.

How to disable the slideshow duration for videos only?

So, if assets are like below, then each duration I want is on the right.
```
1. photo1.jpg      ->  30 sec
2. video1min.mp4   ->  1 min
3. photo2.png      -> 30 sec
4. video2min.mp4   ->  2 min
```

nanpuhaha, 2021-12-21:
#47

closed 2021-12-21 by nanpuhaha

Geektoolkit, 2021-12-21:
Currently this is in a rough state. 2.23 has the fix for this, but it may bring other issues..I've not promoted it as 'released' because I haven't heard enough feedback on testing of it yet to consider it such (and I know there are some playlist bugs from moving to a database lurking in there). 
2.19 is my latest that is reasonably stable, but I don't have the feature in there. Further, video playback is broken and will need to be re-engineered thanks to the Raspberry pi foundation moving away from OMXPlayer. so currently..no good answer. working on it.