Geektoolkit/Dynaframe3 #168: Black screen for images

vatsake, 2021-12-21:
In stable version and in 2.23.

For 10-20 minutes the program works well.
But after that all images in the playlist display just black screen. The videos play fine.

Geektoolkit, 2021-12-21:
Are you using VNC and moving a mouse? If so that crashes the renderer that suppresses the mouse cursor. I've seen this if that happens.

vatsake, 2021-12-21:
I see, thank you for that info.
I'll try without VNC.

vatsake, 2021-12-21:
This time I even disabled VNC.
Now it even started in black screen - for images (desktop background was shown at startup) - although videos did play.

`21/12/2021 16:28:Timer_Tick: Reload settings was true... loading settings
21/12/2021 16:28:KillVideoPlayer - Entering Method.
21/12/2021 16:28:VideoPlayer CheckStatus returning fasle..: Video has exited!
21/12/2021 16:28:GoToNext: called
21/12/2021 16:28:ERROR: GoToNext was unable to find current item in playlist!
21/12/2021 16:28:GetPlayListItems: Clearing item list..
21/12/2021 16:28:GetPlayListItems: Getting playlist Items...
21/12/2021 16:28:GetPlayListItems: Shuffle is off. Sorting items...
21/12/2021 16:28:----------------------Begin Playlist Dump----------------
21/12/2021 16:28:Image:/home/pi/Dynaframe/web/uploads/031717_1.jpg
21/12/2021 16:28:--------Tag dump-------------
21/12/2021 16:28:Title:
21/12/2021 16:28:Artist:
21/12/2021 16:28:Comment:
21/12/2021 16:28:Keywords:
21/12/2021 16:28:--------END END-------------
21/12/2021 16:28:Image:/home/pi/Dynaframe/web/uploads/031717_10.jpg
21/12/2021 16:28:--------Tag dump-------------
21/12/2021 16:28:Title:
..
..
..
:Video:/home/pi/Dynaframe/web/uploads/x.mp4
21/12/2021 16:28:------------------- END PLAYLIST DUMP  Took: 20 ms.  -----------------
21/12/2021 16:28:GoToNext: Current Index Value is: 0
21/12/2021 16:28:CurrentPlayListItem Count is: 104
21/12/2021 16:28:GoToNext: Valid Playlist seems to be detected...Loading next item at index: 1
21/12/2021 16:28:GoToNext: Next Playlist Item will be: /home/pi/Dynaframe/web/uploads/031717_10.jpg
21/12/2021 16:28:Next file is: /home/pi/Dynaframe/web/uploads/031717_10.jpg
21/12/2021 16:28:PlayFile() called with TransitionTime=0 and path: /home/pi/Dynaframe/web/uploads/031717_10.jpg
21/12/2021 16:28:KillVideoPlayer - Entering Method.
21/12/2021 16:28:Media is now set.
21/12/2021 16:28:Dynaframe Initialized...
21/12/2021 16:28:SetupWebServer: Successfully setup webserver on default port :8000
21/12/2021 16:28:Refresh settings was called
21/12/2021 16:28:Current App Settings
21/12/2021 16:28:FadeTransition: 0
21/12/2021 16:28:SlideshowTransitionTime: 5000
21/12/2021 16:28:FontSize: 50
21/12/2021 16:28:FontFamily: Terminal
21/12/2021 16:28:DateTimeFormat: H:mm tt
21/12/2021 16:28:Rotation: 0
21/12/2021 16:28:Shuffle: False
21/12/2021 16:28:ImageStretch: UniformToFill
21/12/2021 16:28:VideoStretch: Fill
21/12/2021 16:28:VideoVolume: True
21/12/2021 16:28:isSyncEnabled: False
21/12/2021 16:28:Number of Sync Clients: 0
21/12/2021 16:28:Current opacity: 1
21/12/2021 16:28:KillVideoPlayer - Entering Method.
21/12/2021 16:28:VideoPlayer CheckStatus returning fasle..: Video has exited!
21/12/2021 16:28:GoToNext: called
21/12/2021 16:28:GoToNext: Current Index Value is: 1
21/12/2021 16:28:CurrentPlayListItem Count is: 104
21/12/2021 16:28:GoToNext: Valid Playlist seems to be detected...Loading next item at index:
`

Geektoolkit, 2021-12-21:
Once you're in that state it may not work till you reboot. Please try that and let me know. The process gets stuck so trying to launch it again without a reboot or killing it will cause that

vatsake, 2021-12-22:
It shows with less images, but for some reason it shows only 1 picture.
In the log it says it changes to another picture, but it still displays the old one.

Videos still work fine.

vatsake, 2021-12-30:
If this issue is only mine, then I'll close it.

closed 2021-12-30 by vatsake