Geektoolkit/Dynaframe3 #175: Changing Port?

abenzick, 2022-01-06:
It looks like some internal networking is blocking port 8000 on my device. Is there a config option to change the port number to something that may be open such as 80?

Geektoolkit, 2022-01-08:
80 is a reserved port which requires sudo to use, but that causes issues. There is a setting in app settings.json I believe to set the port of you parse that, I haven't tested that. 
   Dynaframe pro is the next version, it'll work off of port 80 by default. I had to rewrite a lot to make that work

abenzick, 2022-01-09:
Is settings.json a runtime parsing of the variables or would I have to compile things on my own to get settings.json to load?

Geektoolkit, 2022-01-09:
Appsettings.json is a runtime check so should be able to edit and run. Just make sure you don't mess up the json formatting! I'd also back up the file just in case

abenzick, 2022-01-14:
appsettings.json is jsut a bare frame with test1 and test2. Is there an example anywhere of the setting name for the port? 

Thank you!

abenzick, 2022-01-14:
I did try to a regular "ListenerPort" value in the appsettings.json and I can't tell if it gets parsed. I changed it to "80" for testing and got System.Net.HttpListenerException Permission denied

That made me think it was reading the settting. I then changed it to 8080 and the app ran with no area but the IP display text said port 8000 instead of 8080