Geektoolkit/Dynaframe3 #179: Support Tif file format?

Badlee2020, 2022-01-31:
I am using Dynaframe3, it seems as if only jpg file format can be displayed. Does it als also support other picture formats?

Geektoolkit, 2022-01-31:
It supports bmp, png, and jog formats. It uses Avalonia and skia so it's currently limited to those formats. I'm looking into Tiff support but it's going to be a bit. I'm currently working on animated gifs

raspitakesovertheworld, 2023-09-05:
Support for webp would also be useful, as they are the successor for the always horrible and terrible crappy looking low resolution / low color palette GIFs. It is a shame that Avalonia and skia has such a limited format support. Unfortunately, we always have more and more and more formats, they never become less.