Geektoolkit/Dynaframe3 #184: How to start dynaframe

sajovicd, 2022-04-03:
Hello,
I believe this is newbie question but hovewer... 
I am running on raspberry pi 3B+ and when I want to exit dynaframe I just press ESC and it stops the program.
But how do i start it again?
I tryed with ./Dynaframe directly from command line but with no succes.
Sorry for dumb question and thank you for answering me.

Geektoolkit, 2022-04-03:
Try cd Dynaframe to change to the directory, and then try .\Dynaframe (I may have that slash backwards). Or "sudo bash run.sh" may work as well

Cinderhaze, 2022-08-17:
I'm running into similar issues.  I had to edit run.sh to use my user's home directory (as that script assumes user is named pi).  I still am getting a Seg fault whenever I directly, or the run.sh script indirectly, tries to run ./Dynaframe.

```
~/Dynaframe $ sudo bash run.sh
run.sh: line 23:  1328 Segmentation fault      ./Dynaframe > /home/dawiest/Dynaframe/logs/dynaframe-${now}.log 2>&1
```

I'm also running on a Pi 3B+

Geektoolkit, 2022-08-17:
Yeah the script was written back when pi was the default user. If you're hitting segfaults it's tough to figure out what may be wrong...it could be the first image is too big and it goes out of memory, or it could be some randome dependancy that's changed.  
I've moved over to supporting the patreon release which has a TON of fixes over this codebase, and left this here for developers that are looking to have a codebase to work from (though honestly it hasn't gotten much pickup...people wnat the functionality but not to code it up apparently).

ribbal, 2023-01-06:
Hi, I'm new to Dyanframe.  I haven't installed it yet, and have just been reading through some of the issues.

I see that the script(s) assume the home folder is /home/pi, and users have raised a number issues (I guess) because their user is not called 'pi'.  Can I suggest that the scripts are amended to use the home folder of the current user instead, which is: '~/'.  This should be a simple fix that I would expect may eliminate some of the noise.

ribbal, 2023-01-06:
Ignore me!  I've just spotted this PR: https://github.com/Geektoolkit/Dynaframe3/pull/201, which appears to revolve the problem (though in a different way to the above).

Geektoolkit, 2023-01-07:
The problem is it runs as root and is looking for the pictures folder. Since root doesn't have a pictures folder then it can't find it, so it's been hardcoded to pi. It only runs on Buster (not bullseye) which defaulted to a pi user.

I'm heads down on the Patreon Dynaframe pro, so I've left this source open for others to tweak. I'm accepting PRs :)