Geektoolkit/Dynaframe3 #187: Start Up Issue

ResanS, 2022-05-01:
Hey, I downloaded Dynaframe and put some pics in the folder, but Dynaframe won’t start. I tried the ./Dynaframe command but it says „no such file or directory“

ResanS, 2022-05-01:
cd Dynaframe won’t work either…

AROJINSTIC, 2022-08-20:
You are a great boss, more power to your elbow, it's working ! Question......... At the 12mins of my inserted pictures that slide, my picture stopped and the follow come transition image starts till infinity, except I touch the mouse that continues my inserted pictures and stop again after 12minute, can I remove the follow come transition images & let it display only the image inserted in the picture folder.... Looking forward to seeing your reply boss.

NoneRighteous, 2022-08-27:
I found this issue occurred when running the 64-bit version of Raspbian, only the 32-bit is compatible with the compiled binary downloaded via the Readme instructions.

raspitakesovertheworld, 2023-09-06:
Hmm, Raspian will support 32 Bit too, that should actually work, but there might be the need for the 32 bit version of the libraries installed too.