Geektoolkit/Dynaframe3 #192: Install Script Failure

warbosstodd, 2022-07-03:
The install script is failing when it tries to download the .zip file. 

2022-07-03 09:36:35 (9.66 MB/s) - ‘Dynaframe2.zip.2’ saved [25269628/25269628]

FAILURE! Dynaframe2.zip did not download successfully

lsmithx2, 2022-09-09:
I got the same error happen to me, I have submitted a fix for this to the repository.

ecto1a2003, 2022-09-28:
> I got the same error happen to me, I have submitted a fix for this to the repository.

hi, how do i use this fix?

tehchy123, 2023-03-02:
> I got the same error happen to me, I have submitted a fix for this to the repository.

Where could we find this?

Faramarzian, 2023-05-03:
i used this and it worked sudo wget https://github.com/Geektoolkit/Dynaframe3/releases/download/2.01/install.sh && bash install.sh; rm -f install.sh

Faramarzian, 2023-05-03:
sudo wget https://github.com/Geektoolkit/Dynaframe3/releases/download/2.06/install.sh && bash install.sh; rm -f install.sh   that was the latest version that worked

overhead85, 2023-08-30:
> sudo wget https://github.com/Geektoolkit/Dynaframe3/releases/download/2.06/install.sh && bash install.sh; rm -f install.sh that was the latest version that worked

That fixed the problem for me... update the description on youtube...

raspitakesovertheworld, 2023-09-06:
Is it known why the download of the Zip fails?
I downgraded to 2.06, but wonder what this is not fixed in the newer version.