Geektoolkit/Dynaframe3 #203: Showing a picture even though it was deleted

SpaceFahad, 2022-11-06:
I have an issue with my Dyna frame showing a picture I already deleted from the folders, I tried everything. However, it still shows it even though I made sure it was deleted and I emptied the web/upload + pictures, then uploaded new ones, new ones show alongside the deleted one.

SpaceFahad, 2022-11-09:
> not sure what would do this, have you restarted the unit and have the problem still happening?

I have tried restarting, shutting down for a few minutes, and updating the pi, but none worked. The problem still persists.

qwksilver, 2022-11-14:
i can't replicate this issue, chances are a permissions item has prevented full deletion of the file, as this version is almost 2 years out of active dev now, i would urge you to come check out the patreon edition DynaframePro, it's lightyears ahead of this point, and likely this problem will vanish with the newest image.

SpaceFahad, 2022-11-16:
> i can't replicate this issue, chances are a permissions item has prevented full deletion of the file, as this version is almost 2 years out of active dev now, i would urge you to come check out the [...]

Issue was identified, for some reason a photo I deleted earlier did not get deleted, but instead it was renamed to " .jpg" and was hidden.. I had to show hidden files in WinSCP to be able to find and delete it. 

As for the DFpro, will check it out, thank you!

closed 2022-11-16 by SpaceFahad