Geektoolkit/Dynaframe3 #205: Error first start

MarianoMatute, 2022-11-09:
Hi,
I followed the guide: 
`sudo wget https://github.com/Geektoolkit/Dynaframe3/releases/download/2.19/install.sh && bash install.sh; rm -f install.sh ; sudo reboot` and it work fine.
I'm having this error after initial installation and first run.
" Unhandled exception: System.IO.DirectoryNotFoundException: Could not find special directory 'My Picture' "
I'm running Dynaframe in a Pi 4. 
I've renamed the folder "Pictures" to "My pictures". same error. 
The folder has 3 pictures into.

What could happen?
Thanks.

qwksilver, 2022-11-09:
Be carefull with folder names and capitalization.  so you should have in your pi a directory as follows
`/home/pi/Pictures`  if you want to make another that is fine, but do not remove the stock directory.

if there is still a problem after recreating the default photo directory, then do `sudo rm -rfv ~/Dynaframe`  then rerun the install string above.

I will also let you know that there is a newer version of Dynaframe out there called DynaframePro and it's a patreon supported project, and has advanced a long way from Dynaframe3.  You may want to look there, it's seriously worth the cost of a coffee to hop in.

[Dynaframe Pro Patreon](https://www.patreon.com/Geektoolkit)

MarianoMatute, 2022-11-10:
Hi, Thanks.
I did your steps above and Pictures folder was solved. And now I have a new error that I cannot start the GUI.
"Unhandled exception. System.InvalidOperationException: Default font family name can't be null or empty.
   at Avalonia.Media.FontManager..ctor(IFontManagerImpl platformImpl) in /_/src/Avalonia.Visuals/Media/FontManager.cs:line 28"
I tried to install some fonts manually, following som google search but this is not working. :-(
Regarding the new DynaframePro, I know there is a subscription app, but I would like to test first this app to see if it is what I really need. My goal is to have a Pi with a 24" monitor to show my google photos as a digitalframe.

Thanks.

qwksilver, 2022-11-14:
it looks as though the file /home/pi/Dynaframe/appsettings.json is corrupted.  the reference copy is located in the repo, here: https://github.com/Geektoolkit/Dynaframe3/blob/master/appsettings.json  and also in the zip you use the wget command to obtain.  

The subscription version is lightyears ahead of this, and would be well worth a months try out to grab a recent version, or even to join up on discord and possibly opt in to the testing program for even newer copies.

qwksilver, 2022-12-10:
assuming this is resolved, please comment if you still need help.

closed 2022-12-10 by qwksilver