Geektoolkit/Dynaframe3 #210: os

EQUALIT-CG, 2023-01-13:
Hello @Geektoolkit

Question i see you use Raspberry desktop
can we not just use Raspberry Pi OS Lite

i'm happy to use it but have you tested this ...

Geektoolkit, 2023-01-13:
OS lite doesn't have some of the necessary libraries needed for Avalonia, which this is built on. You can try it to see if it works today, but I suspect it'll crash and code fixes would be needed to get it boostrapped

EQUALIT-CG, 2023-01-14:
Oh unlucky

that would be sick if you could get it written up at some point / worked out i have 0 clue was just asking thanks for the info please open this again if you do end up working it out cheers thanks for this yet to work on something like this

closed 2023-01-14 by EQUALIT-CG