Geektoolkit/Dynaframe3 #219: Script hardcoded for "pi" user

raspitakesovertheworld, 2023-09-06:
The Raspian OS now asks you for a username when installing, I chose another than pi, but noticed that the script seems to rigidly assume that the user will always be called pi.

It might be useful to mention that in the install instructions, to always use the user "pi" for the install, so that it will work.

Geektoolkit, 2023-12-31:
Dynaframe only works correctly on Raspbian Buster, which I believe is mentioned. If you're using a later OS you're going to hit other issues besides the pi username including videos not working at all.  There is a ton of work that would be needed to move this up to work on bookworm and such.  That work is mostly done in Dynaframe pro (we'ver moving it to bookworm in Jan/Feb 2024) but that constant need to update the code to keep it 'working' is one of the reasons it went to a patreon supported model. VLC is another dependancy which changed arbitrarily along with the compositor and other things. Essentially what Dynaframe does is so OS dependant from the audio stack to the video player and such that changes to the OS require quite a bit of work or sometimes entire rearchitectures to keep it 'updated'.

raspitakesovertheworld, 2024-01-17:
In understand the problem of the environment changing so it needs constant changes and adjustments. I would recommend to containerize the whole app, then it is independent of the OS it runs on, as all the dependencies are are satisfied and won't change. For now it is good to know that Buster is required for the proper function, thank you for all the hard work.