Geektoolkit/Dynaframe3 #222: how to run OMXplayer

DiegoLosco, 2023-11-19:
Hi i've installed Dynaframe 2 on a raspPI 4 and OMX player is not installed.
I've manually run the installation on the latest Rasp OS version , and manually change the installation path, but omxplayer is no more available for the installation
but I've installed it via snapD
and it is installed as OMXplayer-pi but it is not reorganized by slideshow
any hint for this issue?

many tsk

Geektoolkit, 2023-11-28:
Dynaframe2 is only supported on Raspbian buster. Beyond that Debian took changes which removed OMXPlayer.  If you'd like to have it work you can edit the source code and compile it or submit a PR to fix it.

Geektoolkit, 2023-12-31:
It's a visual studio project/solution which is included. if you load up VS2023 and load the solution, it should compile 'as is' (it'll pull down the avalonia nuget packages and such).