Geektoolkit/Dynaframe3 #233: Dynaframe don't start

klodespe, 2024-09-05:
Hi, thank you for your response.
I have just installed Dynaframe on RaspberryPi4 end another on RaspberryPi3.
In every case I have used a command in the "Code" page, but when I reboot Dynaframe don't start.

I looked in the log file and this is the sentence:

/home/pi/Dynaframe/run.sh: riga 14: ./Dynaframe: cannot execute: required file not found

How I can resolve?

errorkallu, 2024-09-10:
same for me

Lakiman7, 2024-10-13:
same for me

Lakiman7, 2024-10-13:
I have tried with multiple versions of the OS on both the Raspberry Pi 3B+ and 4B, but the same error appears in the log: ./run.sh: line 23: ./Dynaframe: cannot execute: required file not found.

SitzPatrick, 2025-01-16:
I used Raspberry PI OS (32-Bit ) and seem to work for after getting through the no usable libssl found error

pbassut, 2025-04-07:
Same issue

pbassut, 2025-04-07:
Figured it out.

Seems like it only works with this specific version of RaspiOS: 2020-08-24
Found it [here](https://downloads.raspberrypi.org/raspios_armhf/images/raspios_armhf-2020-08-24/)

Also, has to be flashed with balenaEtcher and not Raspberry Pi Imager. You will then, need a mouse and keyboard.